A-Emile/Lime_Gen3_IoT_Replacement #2: feature request: BT OTA and stack change

Pikokosan, 2023-04-18:
Im looking at the code and it's size. you may wanna look into switch to nimble stack to reduce the program size. also im looking at https://learn.sparkfun.com/tutorials/esp32-ota-updates-over-ble-from-a-react-web-application/all to do firmware updates which would eliminate the need for starting the wifi and having to use that of OTA updating.

Im messing with the nimble stack tonight to see if i can convert it over.

Pikokosan, 2023-04-18:
Got the code compiling with the Nimble stack. need to do further testing but the program size went from 55% to 30% and dynamic memory went from 12% to 11%

Pikokosan, 2023-04-18:
I have put a draft pull request #3  . id recommend starting a new branch called nimble. i can the request the pull to that so it can be tested before going to the main.

A-Emile, 2023-04-19:
Hey bro, thank you very much. Im not very good with github. I have just merged you pull request but it instantly went in the main branch. So i reverted it. Can you tell me, how to do it correctly? And again, thnak you very much!

Pikokosan, 2023-04-19:
This will explain it better than I can.
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-and-deleting-branches-within-your-repository#creating-a-branch

Pikokosan, 2023-04-19:
I'm about to start work on the ota over ble feature. I had tried the spark fun method but ran into connection issues when updating. It would hang and not recover in the web app so I'm looking into this https://github.com/vovagorodok/ArduinoBleOTA

It's even got a flutter app so that makes it easier to implement into the limer app.

A-Emile, 2023-04-20:
> I'm about to start work on the ota over ble feature. I had tried the spark fun method but ran into connection issues when updating. It would hang and not recover in the web app so I'm looking into [...]

Thats really nice! I will try to get the github stuff working this evening..

Pikokosan, 2023-04-26:
Designed a small test board and ordered. 

Got the arduinoOtaBle library implemented. Haven't be able to test fully due the libraries app being broken.

Pikokosan, 2023-05-06:
All done and bench tested. Dev pcb came in. Firmware seems to be sorta working. Test setup doesn't have the rear motor so the controller is flashing a error. I'm gonna have to find my old hard drive with my notes for the gen2.5 decoding. 

I had figured out throttle position and brake position and error codes

A-Emile, 2023-05-08:
Respect! Sounds really nice! I was very busy the past weeks so i still wasnt able to get the github stuff working. But i'll dig into it in the following days..
Can you show us some pics of the pcb you made? Are you going to open source it, so everyone can create it?

Pikokosan, 2023-05-09:
It's really just a dev board right now. I for sure plan to keep this whole thing open source
[image]
[image]
[image]

Pikokosan, 2023-05-09:
[image]

Pikokosan, 2023-05-09:
It's alive and runs now. Gonna try to design a 3d printable case. This is still just a dev board. Touch sensor and the speaker aren't in this design and I got the silk screen wrong for the wiring. 
The good:
The nimble branch I started now has working ble ota updating functioning. 

The bad:
It's not currently reading data from the controller so no battery info or speed. I can see the last packet in the debug charteristics and it does show the battery level.

Ota does require changing a build flag and using my fork to work nicely with the crc library. (Currently working with the original author of the ota library on getting my code pulled to the main branch.

Pikokosan, 2023-05-09:
I'll also be trying this on my 2.5 scooter in the next week. 2.5 scooter have 1 connector instead of two are were much easier to mod.

aIecxs, 2023-06-08:
take note the LM2596S max. input voltage is 35 volt, while the charger input is 42 volt. better go for LM2596HVS or MP4560 module. the connector is Hirose DF13EA-20.

RSF99, 2023-06-18:
Hello guys, hope everything is going well to everyone. So i read everything probably on how to make a personal lime, i flashed 3 controllers but i always have the same problem it works for a minute and locks (shows scan to unlock on the screen) and it's the same thing every time i reflash it. 

Tried plugin the charger when it locks but first 2 attempts it unlocks but the attempts after that it show charging on screen and scooter is locked (shows scan to unlock on the screen).
- i still have the loT '' black box'' without a sim ofc, but i see that it's useless now since it can't be flashed.
- i read A-Emile project post and i was really mind blown of the great work you've put in this project. 
- so i bought an ESP32 DEVKIT V1, buck converter and ams1117 to have a stable 5v as you mentioned. 
- and now i saw that Pikokosan have successfully made a personal lime and I'm very happy for you brother hard work pays off finally. 

!!!!!!!!!!!!!!

NOW i need help with somethings:

*1/ How do i connect the ESP32 with the buck converter and ams1117 to the scooter PLZZZ?
*2/ Wich file do i need, i saw on A-Emile post ( Flash the arduino code from LimeIoT folder to the esp32) but there's 16 files in the LimeloT folder wich one do i need or is there a complete file that you can send to me in private PLZZZ? 
*3/ How do i flash the ESP32 (the procedure explained) plzz?
*4/ Pikokosan i see that u have the esp32 installed in some pcb, how did you do it plz and what do i need to connect on its back exactly in wich order.
*5/ Pikokosan i guess you flashed the esp32 with your custom arduino code am i right? If yes could you send it to me 
*6/ Can i use the scooter temporary with just the controller? Without the loT

Please A-Emile and Pikokosan Help me to successfully convert this to a personal scooter i work far from my home and i go by foot every morning and go back by foot every night please guys helpp me, just tell me what to do and I'll do it 🙏

RSF99, 2023-06-18:
[image]

RSF99, 2023-06-18:
[image]

RSF99, 2023-06-18:
[image]

RSF99, 2023-06-18:
Alecxs i see that you have knowledge about it please guys if you can help me i would really appreciate it, I'm really frustrated been trying my best but no success 🙏

aIecxs, 2023-06-18:
github is the wrong place to ask. find me on scootertalk.

RSF99, 2023-06-18:
Ohh sorry for that, went everywhere on the internet searching how.

How can i find you on scootertalk?

RSF99, 2023-06-19:
> github is the wrong place to ask. find me on scootertalk.

Can you give me a way to contact you I'm really sure that i won't take much from your time, man i only have a phone, I've been borrowing a friend laptop to do things i really need some help plz

RSF99, 2023-06-23:
Ok so i got the needed hardware, i took @aIecxs advice on using a buck converter with LM2596HVS since the LM2596S max input voltage is 35V but the charger has a 42v input. 

**I have two questions :**

- 1. How do i wire the (Controller 36v cable) to the	(Buck converter) to the (Ams1117) to the (Vin pin) on the esp32.
I would really appreciate if you could make a diagram of how is everything connected please? Or at least tell me how they're all connected (port to port) 

- 2. Do i connect also the (display 5v cable) to the same (Vin pin)?

- 3. Should i connect the rest of the wires to the corresponding pins as @A-Emile said? (as shown on the Table in the picture ) : since i don't have the same version of esp32 as he has (picture of my esp32 attached)
 
[image]

- 4. @Pikokosan could you please share the Pcb design that you have made. 
And could you post any updates you have about the nimble branch that you started working on and if you did any modifications on the scooter's design? 

 **Hardware :**

**- Esp32 Devkit V1 with 30 Pins :**

[image]

**- Buck converter with LM2596HVS :**

- - *Output voltage 3V -> 30V
- - *Input voltage 4.5 ->  50V
- - *Current output max. 2A

[image]

**- Ams1117 :**

[image]

[image]

RSF99, 2023-06-30:
**Update :**

- Flashed the esp32 with the arduino code from the LimeIot folder. 

- Wired everything as @A-Emile said. (didn't use a Ams1117 just used the buck converter) 

- I've put the esp32 and the buck converter in the *plastic box"

- connected the plastic box to the scooter.

**Result :**

- Front Light is flashing the minute i connect the plastic box

- LED under the screen is red 

- screen shows : scan to unlock and battery 0% (I charged the scooter when the original black box was connected and it was 98% ) 

- i download the android app, i scan for scooters and my scooter appear but when i press connect it doesn't connect
__________________________________________________________________

My conclusion is either the wiring is off or the code has some errors i guess. 

What do you think guys? @A-Emile @reuzurq @Pikokosan

**Here's some pics :** 

- The scooter :

https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/assets/136793581/d616c7a1-ae7a-4a8d-ac24-10e9f06c1946

- The plastic box (esp32 + buck converter) 

[image]

[image]

[image]

[image]

[image]

[image]

- The wiring of the esp32 :

[image]

[image]

RSF99, 2023-06-30:
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/assets/136793581/f57f9ca0-caf7-4db1-a545-f3d1cc253c1c

insestito36, 2023-09-18:
> It's really just a dev board right now. I for sure plan to keep this whole thing open source [image] [image] [image]

Can you share the files for make my own ? Thanks a lot for your effort!

RSF99, 2023-09-19:
@A-Emile @aIecxs @Pikokosan Hello brothers , i hope everything is going well with your lives :) 

it's been a while now since thanks to you, I've been going to work and university comfortably without spending grocery money on Taxi's and Uber's , added weight even lol . 

and here's the but , so i was going as usual to the supermarket and was waiting on the Crossing Lines to turn green to pass and "you know who" was next to me and saw the " Scan to unlock " and the " Lime " Logo on the display and asked many many questions while my heart was popin like a popcorn.

i know it goes away and the speedometer and battery percentage comes once you cut the 5v to the esp32 and put it back ( i did that for a few days but i got tired and i soldered a button to the vin from the 5v ) and that i can't use the app to turn it on and off .

straight to the point , is there a way to remove any Lime or scan to unlock from the display by modifying the code and leave it blank or write anything in it's place pleaseee !

and many many Thanks for the Answers in advance and just not ignoring me :)