A-Emile/Lime_Gen3_IoT_Replacement #6: Info: commands and packet format

Pikokosan, 2023-05-07:
Here is some more info i learned when modding the lime 2.5. looks like the packets haven't changed much.

 keep alive:           0x4643110100084C494D4542494B45BE8A 
 light on: 				0x464316120001F12B26
 Show firmware version: 0x464311140000EAFB 
 Show UART version:     0x464311150000DDCB 
 Show hardware version: 0x4643111800009F9A
 Show some weird id:    0x464311200000F39E 
 Blink front light:     0x464316130001F15D92
 Blink front light stop: 0x46431613000100A2AC
 Lock motor wheel:      0x464316630001F11FE7
 Unlock motor wheel:    0x464316630001F00FC6
 turn off and lock:     0x464316610000F0D19F
 turn on and unlock:    0x464316610000F1C1BE

example response from controller
`46 58 0C FF 00 27 11 00 00 40 00 00 41 53 6A 42 00 7B 44 4F 52 4F 61 F0 80 01 40 72 07 07 01 77 82 00 00 00 00 E0 00 0A 93 F1 45 00 00 1E BD`

byte 1 is the start byte
byte 2-6 are part of the header
in the example byte 7 = 0x27 which in dec is 39 which is the length of the frame meaning 6(SOF and header)+39+2(the crc) is the full packet.

HenryCavil, 2023-05-07:
How i can  eavesdrop on communication between the gps and the controller?

Pikokosan, 2023-05-08:
Cheap logic analyzer is the easiest way. Another way would be with an arduino mega and using it as a passthrough

Pikokosan, 2023-05-12:
Did some more decompiling and found it uses stm32flash on the iot to program the motor controller. looking for the command to put it into boot loader mode right now. 

i can see it puts it into the bootload then call

 ```
stm32flash %s -k","/dev/ttyHSL3
 stm32flash %s -u","/dev/ttyHSL3
 stm32flash %s -w %s","/dev/ttyHSL3
```

RSF99, 2023-06-18:
Hello guys, hope everything is going well to everyone. So i read everything probably on how to make a personal lime, i flashed 3 controllers but i always have the same problem it works for a minute and locks (shows scan to unlock on the screen) and it's the same thing every time i reflash it.

Tried plugin the charger when it locks but first 2 attempts it unlocks but the attempts after that it show charging on screen and scooter is locked (shows scan to unlock on the screen).

i still have the loT '' black box'' without a sim ofc, but i see that it's useless now since it can't be flashed.
i read A-Emile project post and i was really mind blown of the great work you've put in this project.
so i bought an ESP32 DEVKIT V1, buck converter and ams1117 to have a stable 5v as you mentioned.
and now i saw that Pikokosan have successfully made a personal lime and I'm very happy for you brother hard work pays off finally.
!!!!!!!!!!!!!!

NOW i need help with somethings:

*1/ How do i connect the ESP32 with the buck converter and ams1117 to the scooter PLZZZ?
*2/ Wich file do i need, i saw on A-Emile post ( Flash the arduino code from LimeIoT folder to the esp32) but there's 16 files in the LimeloT folder wich one do i need or is there a complete file that you can send to me in private PLZZZ?
*3/ How do i flash the ESP32 (the procedure explained) plzz?
*4/ Pikokosan i see that u have the esp32 installed in some pcb, how did you do it plz and what do i need to connect on its back exactly in wich order.
*5/ Pikokosan i guess you flashed the esp32 with your custom arduino code am i right? If yes could you send it to me
*6/ Can i use the scooter temporary with just the controller? Without the loT

Please A-Emile and Pikokosan Help me to successfully convert this to a personal scooter i work far from my home and i go by foot every morning and go back by foot every night please guys helpp me, just tell me what to do and I'll do it 🙏 sorry for commenting in both places but I'm worried that you guys don't receive my comment or don't read it :/

RSF99, 2023-06-18:
[image]
[image]

HenryCavil, 2023-07-09:
Sj2.5 controller have freescale mke02z64 chip. Any ideas how to read and flash firmware? I have one motor controller with working commands firmware.

reuzurq, 2024-01-02:
> Sj2.5 controller have freescale mke02z64 chip. Any ideas how to read and flash firmware? I have one motor controller with working commands firmware.

You can flash ke02 chips with [OpenOCD](https://github.com/openocd-org/openocd/) using [ST-LINK](https://www.amazon.com/s?k=ST-Link+V2)