A-Emile/Lime_Gen3_IoT_Replacement #7: Max speed

HenryCavil, 2023-05-07:
DE market have 20kmh limit. Unlocked firmware have 27kmh. Maybe there is chance to change max speed of the scooter above this 27kmh?

A-Emile, 2023-05-08:
Yeah, it should be possible. We just have to find the command. Maybe we can try bruteforcing..?

Pikokosan, 2023-05-08:
It's a internal limit in the firmware. There won't be a command to change the speed limit

A-Emile, 2023-05-08:
Yeah, but i think it could be possible to change it over uart because in the main binary of the iot are a few functions that looks like they change the max speed based on location. And I just also heared, that the controller is made by lishui and parameters may can be set by a software called "Lishui Controller FOC parameters adjuster" but i wasnt be able to connect it to my ftdi programmer.
It could also be possible to flash the following open source firmware: https://github.com/EBiCS/EBiCS_Firmware

A-Emile, 2023-05-08:
I also found this thread, wich could be interesting: https://www.pedelecforum.de/forum/index.php?threads/bagier-sinus-controller-programmierung.25904/page-4#post-756210

Pikokosan, 2023-05-09:
I noticed that too when I was working on the 2.5 controller. Never could get it to talk to the software. From the binary I pulled from the 2.5 it looks like a custome flavor. Maybe the white wire is used to enable a bootloader or put it into a configuration code?

A-Emile, 2023-05-09:
Im not sure but i think the white wire is ntc

A-Emile, 2023-05-09:
And it think the problem with the software is, that its made fur older uart adapters. Since it doesnt even try to speak with my ftdi. There is a newer version of it wich supports ftdi programmers. But it isnt available to download anymore. Maybe someone in the forum still has it..

A-Emile, 2023-05-09:
I just had a thought: there is a max speed screen for the display. Maybe there are zones in the city, where speed is even more limited like to 10kmh. We can sniff one and drive in and out such a zone...

Pikokosan, 2023-05-09:
> Im not sure but i think the white wire is ntc

That makes more since. I just looked at the 2.5 version and it doesn't have that.

MrKris7100, 2023-11-29:
Maxiumum speed is limited by the motor. It's maked in spacific way to have maximum speed of ~30km/h. If you wanna increase maximum speed you need to apply higher voltage to motor or replace motor. I tested 48V on this mottor with generic controller and it's driving around 47km/h.

reuzurq, 2024-01-02:
> There won't be a command to change the speed limit

There is a speed limit variable, u can edit it both in fw binary and via serial commands. Stock esc can go up to 34kmph (similar to ximi pro2) when fully charged.

csizmard, 2024-01-30:
Hi, I give some code for exact speed setting for the Lishui controller (Lime 2.5)

Can u help me how to work around switch off the only by "pedal assist" starting?
I dont find your know commands, where are theys? 

I thinking of the fw all parameters readout command finding actually... 

Code  | Km/h
4643161100027604C6 | 7,2
464316110002EE0677 | 10,4
464316110002166860 | 12,6
4643161100021008A6 | 13,5
464316110002C5937E | 15,8
464316110002010AB6 | 16,8
464316110002880A17 | 17,1
464316110002320C86 | 18,9
464316110002AA0E37 | 22,7
464316110002DC1066 | 26,4
464316110002CD1276 | 33,2
464316110002023AD5 | 33,7 (40-50 alternatives also there for same speed)

reuzurq, 2024-01-30:
Commands you have provided are a bit wrong. Speed governor needs 2 bytes set. Your commands have specified length of 2 but you are passing only 1 byte and checksum (-2) so these commands would fail.

```
4643 16 1100 02 02 ?? | 3A D5
HEAD RW ADDR LN DATA2 | CRC16

?? Missing byte
```

> Can u help me how to work around switch off the only by "pedal assist" starting?

I dont think that this is possible on those - dont have time rn to test rest of the addresses that ive found.

Also 2.5 and 3.0 controllers are a bit different - having some functions mapped to same addresses, an lots of other - unique to the model version.

mciszdaniel, 2024-01-30:
Commands yesterday night I try all, and working (and more 255) on that address.

mciszdaniel, 2024-01-31:
> Commands you have provided are a bit wrong. Speed governor needs 2 bytes set. Your commands have specified length of 2 but you are passing only 1 byte and checksum (-2) so these commands would [...]

The value could be entered with decimal precision, but it accepts it without I think!

reuzurq, 2024-02-01:
> The value could be entered with decimal precision, but it accepts it without I think!

From my observations only the 2nd byte changed the speed, first one did nothing for me but it required two of them.

mciszdaniel, 2024-02-01:
If u send me your codes, i will share mines ;) 

mciszdaniel@gmail.com

reuzurq, 2024-02-01:
> If u send me your codes, i will share mines ;)

Im not very interested.

mciszdaniel, 2024-02-01:
👍🏼

reuzurq, 2024-02-01:
I mean, **if you want to share something with the community - just do it, thats the point of this website.** Dont see any point of this "private exchange" here :3

mciszdaniel, 2024-02-14:
Corrected speed codes:
Requested back each, its echo the complete..

Hex Code | Speed | Hex Data value
4643161100027604C600 | 7,2 Km/h | 7604
464316110002EE067700 | 10,4 Km/h | EE06
46431611000216686000 | 12,6 Km/h | 1668
4643161100021008A600 | 13,5 Km/h | 1008
464316110002C5937E00 | 15,8 Km/h | C593
464316110002010AB600 | 16,8 Km/h | 010A
464316110002880A1700 | 17,1 Km/h | 880A
464316110002320C8600 | 18,9 Km/h | 320C
464316110002AA0E3700 | 22,7 Km/h | AA0E
464316110002DC106600 | 26,4 Km/h | DC10
4643161100020111155A | 28,3 Km/h IOT>CTRL : Set speed limit | 0111
464316110002CD127600 | 33,2 Km/h | CD12
464316110002023AD500 | 34,8 Km/h | 023A

What could be the data conversion? I cant figure out..