A-Emile/Lime_Gen3_IoT_Replacement #11: I got a small problem i see here that you're using esp32 that has a 5v pin but i got the esp32 devkit v1 with 30 pin and it has only a 3.3v pin. 

The question here is :

i need to connect the 5v controller and 5v display to the esp32 5v pin how can i resolve that in my case using a esp32 devkit v1 that has only a 3.3v pin

PS : searched online about the VIN pin and found some vague information (some people say it works for 5v out and some people say it fried there esp32 and smoke came out).

***Some help please, i read almost everything on scootertalk and found that this project is the best one to convert a lime-s gen 3 and has a complete guide on how to do it but i just need some guidance on some small things that I don't understand fully, I'm trying to learn and I have learned alot since the beginning of this journey thanks to you and your knowledge guys 🙏

RSF99, 2023-06-19:
The one you're using ''i guess'' :

[image]

The one I have right now :

[image]

RSF99, 2023-06-19:
I got a small problem i see here that you're using esp32 that has a 5v pin but i got the esp32 devkit v1 with 30 pin and it has only a 3.3v pin.   

The question here is :  

i need to connect the 5v controller and 5v display to the esp32 5v pin how can i resolve that in my case using a esp32 devkit v1 that has only a 3.3v pin  PS : searched online about the VIN pin and found some vague information (some people say it works for 5v out and some people say it fried there esp32 and smoke came out).  

***Some help please, i read almost everything on scootertalk and found that this project is the best one to convert a lime-s gen 3 and has a complete guide on how to do it but i just need some guidance on some small things that I don't understand fully, I'm trying to learn and I have learned alot since the beginning of this journey thanks to you and your knowledge guys 🙏

reuzurq, 2023-06-21:
I am using the same esp as you, in mine - 5V from the buck converter goes into VIN pin on esp32.

VIN (Voltage in) is connected to the onboard voltage regulator that steps down the voltage to 3.3v.
If you are worried about frying your esp32, you can use external voltage regulators.

next time use shorter titles for issues ;)

RSF99, 2023-06-22:
Thank you for your help brother, i really appreciate it. And sorry for the long title I'm new here :).

I have a couple of questions if you don't mind since you're using a setup with the same esp32 that i have.

- are you using an external voltage regulator? or did you connect the buck converter and the display 5v to the vin pin on the esp32 directly? 

- so as i understand i need to connect the wire of the controller 36v to a buck converter and then connect the buck converter to the ams1117 and then connect the ams1117 to the VIN pin on the esp32 and also connect the wire of the display 5v to the same VIN pin? 

- did you connect the wires to the corresponding pins as the A-Emile said? (as shown in the picture) :

[image]

- wich arduino code did you use to flash the esp32 in wich order ( i saw multiple docs on the LimeIoT folder and don't know wich one needs to be used ) 

I'd really appreciate that you help into doing this brother and finishing this scooter setup plz 🙏

RSF99, 2023-06-24:
Hello everyone so i flashed the **esp32 devkit v1** with the arduino code from the LimeIot folder and I'm ready to connect it to the scooter. 

**Could you guys check if my wiring is correct? **

@A-Emile said to connect the corresponding wires to the corresponding pins on the esp32 ( as shown on this picture): 

[image]

- **From the controller cable (with black connector) : **

- Controller 36v (Red wire) to Vin pin 
- Controller Gnd (Black wire) to Gnd pin
- Controller Lock (blue wire) to GPIO 12
- Controller RX (Green wire) to GPIO 1
- Controller TX (Yellow wire) to GPIO 3

(there's a white wire we don't need) 

Ps : for the controller 36v (buck converter + ams1117) to have 5v

- **From the display cable (with blue connector) : **

- Display 5V (Red wire) to 5v Vin pin
- Display Gnd (black wire) to Gnd pin
- Display RX (green wire) to GPIO 17
- Display TX (blue wire) to GPIO 16
__________________________________________________________________

**- ***But when i see on the esp32 devkit v1 pins i find that : **

- GPIO 1 corresponds to TX0 
- GPIO 3 corresponds to RX0
- GPIO 17 corresponds to TX2
- GPIO 16 corresponds to RX2

[image]

**So is everything ok if i wire it as @A-Emile said or should i switch the Wires?**

- **This is wired as @A-Emile said : **

[image]

[image]

**Is this wiring correct please? **

A-Emile, 2023-06-24:
Yess, good job! Wiring looks good.

RSF99, 2023-06-24:
@A-Emile Thank you very much brother i really appreciate all the work you did, god bless you bro you helped me a lot 🙏

This going to be my last question and after the I'll post all of my My project 

How do i connect the controller 36v (Red wire) to the buck converter to the Ams1117 to the 5v Vin pin?? Please 

And the display 5v goes to the same Vin pin yeah so i connect both to the same pin

A-Emile, 2023-06-24:
Yeah, sorry for not responding so long. 39v and gnd of controller goes into buck converter input. Buck converter output to ams input.

A-Emile, 2023-06-24:
And i think i messed some things up in the arduino code in the repo. So dont wonder if some things doesnt work. Or maybe nothing haha. Will update repo later..

RSF99, 2023-06-24:
[image]

@A-Emile Is this wiring correct?

And if it's correct, there's no gnd out from the Ams1117, how can wire the controller gnd to the esp32 gnd?

RSF99, 2023-06-24:
Haha well i need to connect it to the scooter first and then I'll test if everything works or nothing. 
I'm waiting by the minute for your updates brother keep the good work. 
keeping hopes up 🙏

A-Emile, 2023-06-24:
Connect the esp gnd to the gnd of ams and buck converter

RSF99, 2023-06-24:
Alright thank you brother I'm wiring everything now I'll keep you updated.

RSF99, 2023-06-24:
> Connect the esp gnd to the gnd of ams and buck converter

@A-Emile please should i put the output voltage of the buck converter to 3.3v since the Ams1117 is 3.3v or should i put 5v?

RSF99, 2023-06-30:
**Update :**

- Flashed the esp32 with the arduino code from the LimeIot folder. 

- Wired everything as @A-Emile said. (didn't use a Ams1117 just used the buck converter) 

- I've put the esp32 and the buck converter in the *plastic box"

- connected the plastic box to the scooter.

**Result :**

- Front Light is flashing the minute i connect the plastic box

- LED under the screen is red 

- screen shows : scan to unlock and battery 0% (I charged the scooter when the original black box was connected and it was 98% ) 

- i download the android app, i scan for scooters and my scooter appear but when i press connect it doesn't connect
__________________________________________________________________

My conclusion is either the wiring is off or the code has some errors i guess. 

What do you think guys? @A-Emile @reuzurq @Pikokosan

**Here's some pics :** 

- The scooter :

https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/assets/136793581/d616c7a1-ae7a-4a8d-ac24-10e9f06c1946

- The plastic box (esp32 + buck converter) 

[image]

[image]

[image]

[image]

[image]

[image]

- The wiring of the esp32 :

[image]

[image]

A-Emile, 2023-06-30:
The code has errors

A-Emile, 2023-06-30:
Here is the latest working code:
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/releases/tag/latest

closed 2023-06-30 by A-Emile

RSF99, 2023-06-30:
You're the man, thank you brother 🙏

SevkiDeJetskie, 2023-08-19:
For me, the display only lights up the lime logo. As soon as I connect the esp32 to the pc, the scooter turns on and shows 0% battery. I tried it with both esp32 exactly the same.

reuzurq, 2023-08-19:
> For me, the display only lights up the lime logo and the scooter is not displayed in the Bluetooth app. As soon as I connect the esp32 to the pc, the scooter turns on and shows 0% battery.

Check that esp32 is getting enough voltage from your buck converter.