A-Emile/Lime_Gen3_IoT_Replacement #13: Wiring for BUCK Converter?

insestito36, 2023-09-18:
Hello! First of all, thank you for all your efforts making this repository for make personal LIME gen 3.
Can you write how to wiring the BUCK converter? Is this "wiring diagram" correct?

PIN 4 (Controller 36V)----------------      ( In+                         Out+   ) ------- (Input                       OUT 5V )  -- ESP32 ? 
                                                                 (       BUCK                           )            (           AMS1117                  )
                                                                 (       CONVERTER                )            (                                             )
PIN 6 (Controller GND)---------------      ( In-                              Out-) --------(GND                            GND )  --- ESP32 ?

insestito36, 2023-09-18:
And when i open LIMER apk it says "Authorize the flutterreactiveble example app to use bluetooth and location" What this mean?

RSF99, 2023-09-19:
check the other comments

aIecxs, 2023-09-20:
Respect the Code of Conduct

GitHub issues are primarily for discussing code-related topics, such as bug reports, feature requests, and improvements to the project. If you have a general question or need help troubleshooting, it's best to look for more appropriate channels, such as forums or dedicated community spaces.

Recognize that GitHub issues are a resource for project maintainers to manage and prioritize work. Posting non-code-related questions or comments in issues will create noise and make it difficult for maintainers to focus on actual development tasks.

(comment can be deleted)

closed 2023-09-21 by insestito36