A-Emile/Lime_Gen3_IoT_Replacement #15: Throttle not working, battery shows 0%.

MrKris7100, 2024-01-13:
Hello i succesfully flaashed original controler with hacked firmware. I wired up ESP with the scotter and uploaded your code. Scooter is powering up then i can connect to it using your app normally. When i am clicking unlock button the scooter unlocks, shows 0km/hs. And now there is issue. Scooter's display shows 0% on battery. I measured battery with multimeter and it's fully charged. Second problem is scooter doesn't react for the throttle and because of this it's not driving. Strange thing is when i unlock scooter the breaking light blinks a few times then stops and start blinking again. When i lock the scooter or disconnect it lock's itself normally and breaking light stops blinking.

Any advice about this issue? This scooter version is 1.04 (IoT PCB). I tried it with hacked firmware and original IoT with soldered jumper cable to make it boot but it behaviors simillar. Diffrence is it showing battery percentage correctly, is all time unlocked, breaking light reacts to break handle. But still doesn't react for the throttle.

Maybe i did something wrong in the wiring because i disconnected controller and take it to the home i had no IoT plug and just plugged ESP directly to controller but there are two sets of cables coming from controller with same collors. One is with 5 wires, second one is with 6. Maybe i taked wrong cable to connect RX and TX to the ESP? Does it matter?

[image]
[image]
 Some pictures for addition.

A-Emile, 2024-01-19:
Have you flashed the arduino fw i provided here: https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/releases/tag/latest ?

MrKris7100, 2024-01-27:
Well, actually i flashed code just from master branch not from release. I have ESP32 Wroom i guess it doesn't have shock sensor so i should flash version without sleep? Those break light flashes are from triggered alarm?

A-Emile, 2024-01-29:
Yep thats the Problem probably. Try flashing the one without sleep. It should Work..

FrezzZv2, 2024-08-24:
Did you ever figure it out? I'm facing the same issue right now, except I'm already using the "without sleep" fw. One of my scooters works perfectly, but the second one shows 0% baterry 0kmh & every light is off + throttle does nothing.

G-Redos, 2024-09-19:
Try flashing the silver controller next to the battery.
There are 2 versions, 20km and 25km files.

gc9n, 2024-10-17:
> Try flashing the silver controller next to the battery. There are 2 versions, 20km and 25km files.

The lock
Or the unlock ones ?

aIecxs, 2024-10-23:
The 3.3v HIGH Level is maybe not high enough for 5v HIGH level, so behavior is undefined. This may help to understand the problem.
[image]
[How to interface a 3.3V output to a 5V input](https://next-hack.com/index.php/2020/02/15/how-to-interface-a-3-3v-output-to-a-5v-input)