A-Emile/Lime_Gen3_IoT_Replacement #16: Error in Commands.ino  when trying to verify the code

ZRam69, 2024-06-04:
Hi, I'm very much a noob to this so please forgive me if this is obvious.

Not sure what info you need to help me, but I'm running IDE 2.3.2 on a Win 10 64bit V 22H2

My ESP32 Dev Mod is installed and I can upload working test sketchs to it.

The compiling error I get is the following:

C:\Users\ZRam\OneDrive\Documents\Arduino\LimeIoT\commands.ino: In member function 'virtual void MainBLECallback::onWrite(BLECharacteristic*)':
C:\Users\ZRam\OneDrive\Documents\Arduino\LimeIoT\commands.ino:3:50: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
    3 |     std::string value = pCharacteristic->getValue();
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~^~
C:\Users\ZRam\OneDrive\Documents\Arduino\LimeIoT\settings.ino: In member function 'virtual void SettingsBLECallback::onWrite(BLECharacteristic*)':
C:\Users\ZRam\OneDrive\Documents\Arduino\LimeIoT\settings.ino:4:50: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
    4 |     std::string value = pCharacteristic->getValue();
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~^~

exit status 1

Compilation error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested

I've tried to work this out myself, but I'm at a loss, please help!

Thanks in advance, Aidan.

aIecxs, 2024-07-14:
Why don't you use Arduino IDE 1.8.19 what the code is made for instead?

change this type `std::string` to `String` in line 3
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/blob/f7ef6bc/LimeIoT/commands.ino#L3

refer to
https://github.com/espressif/arduino-esp32/pull/8724

ZRam69, 2024-07-15:
Thanks Mate, changing it to String did the trick!

Appresate it!!