A-Emile/Lime_Gen3_IoT_Replacement #20: can i use an esp 32 s2 mini

Valentin09hacker, 2024-09-09:
so the only esp 32 that i have is my esp 32 s2 mini and i am asking if i can use it and which pins do i need to use and if i cant use my esp 32 s2 mini which esp 32 do i  need to get

G-Redos, 2024-09-19:
Does your ESP32 have a 5V GPIO?
There are multiple versions of the ESP32, and many do not have the 5V GPIO.
You need to go over the ESP32 board and make sure you have the 5V GPIO.

RazvanRex, 2024-09-19:
The esp32 s2 mini doesnt even have bluetooth to begin with…
 what does the 5v even have to do with all of this??

G-Redos, 2024-09-19:
The original plans are with using a buck converter to an ams1117 chip.
the ams1117 provides 5V stable current, to connect to the ESP32.

G-Redos, 2024-09-19:
[image]

Valentin09hacker, 2024-09-19:
So now i got an esp 32 wroom and i made the wiring like the diagram and i
think that it doesn't communicate with the controller can you send me your
Instagram so we can talk from there

RazvanRex, 2024-09-19:
I still don’t get where you are trying to get with this.
You can very simply use an esp32 that uses 3v3. The ams1117 is not even necessary if you use a high enough quality stepdown converter…

G-Redos, 2024-09-19:
Yes, you are correct.
I followed the original instructions, to add USB charging ports since they require 5V.

reuzurq, 2024-09-19:
> so the only esp 32 that i have is my esp 32 s2 mini and i am asking if i can use it and which pins do i need to use and if i cant use my esp 32 s2 mini which esp 32 do i need to get

S2 Mini **does not have BLE capability - so it wont work** without redesigning the code to work with WiFi only.

Also please dont create empty & unrelated issues like #19 < here.

Valentin09hacker, 2025-02-26:
I was wondering if it's possible to use any Arduino modules with the scooter just to send the commands to the controller of the scooter to be unlocked forever, because I just want to use the scooter without any other things like phone connection