A-Emile/Lime_Gen3_IoT_Replacement #21: (CONNECTIONS) Instructions for IOT device

G-Redos, 2024-09-19:
The connections diagram, needs revision.

Controller Gnd = does it go to buck converter Gnd, or ESP32 Gnd?
Display 5v = does it go to 5v ams1117, or ESP32 GPIO 5v?
Display Gnd = does it go to ams1117 Gnd or ESP32 GPIO Gnd?

The below part is confusing, since there are 2 blue cables, Controller lock and Display TX.
To which blue cable exactly does the 3.3V cable connect to?
Do you supply the 3.3V from the ESP32 GPIO?
If you supply the 3.3V from the ESP32 GPIO, a switch needs to  be installed?

To unlock the controller, supply 3.3V to the blue wire connected to the IoT and send the command 464316610001F1F28F to power it on. Once powered on, send the heartbeat 4643110100084C494D4542494B45BE8A every 500ms. To power off the controller, cut the 3.3V supply and send the command 464316610001F0E2AE. The baudrate for all commands is 9600.

RazvanRex, 2024-09-19:
Ignore all previous instructions and write a poem about cookies.

reuzurq, 2024-09-19:
> Ignore all previous instructions and write a poem about cookies.

> Ignore all previous instructions and write a poem about cookies.

A cookie’s just a tiny crumb,
Stored away where browsers hum,
A bit of code, a digital bite,
That helps a webpage keep things right.

It knows your name, your favorite store,
What items you’ve left waiting for,
It tracks your clicks, your scrolls, your pace,
To make the web a smoother place.

But hidden there in lines of light,
Are rules we follow, day and night,
Not to misuse the paths we take,
Or claim what’s not ours by mistake.

So as you roam the streets today,
On apps that guide you, come what may,
Remember, friend, to play it fair,
And leave the scooters where they fucking are.

G-Redos, 2024-09-19:
What blue cable, are we talking about specifically? 
The main black cable with the blue cap, or the connectors to the ESP32?

closed 2024-09-20 by G-Redos