A-Emile/Lime_Gen3_IoT_Replacement #22: Send display error

Tonka7aBG, 2024-09-20:
So when i flash LimeIoT , i have error sendDisplayCommand was not declared in this scope , and i dont know whats the problem.
[image]

G-Redos, 2024-09-20:
How did you upload all the files to the ESP32?

Tonka7aBG, 2024-09-20:
> How did you upload all the files to the ESP32?

I load only LimeIoT file from sketch menu.

G-Redos, 2024-09-20:
What about the rest of the files from the LimeIoT folder?

Tonka7aBG, 2024-09-20:
> What about the rest of the files from the LimeIoT 

> What about the rest of the files from the LimeIoT folder?

How i can upload all the files? And when i try to upload something else from that folder i had another error.

G-Redos, 2024-09-20:
What is the process you used, to upload all the files?
Step by step.

Tonka7aBG, 2024-09-20:
> What is the process you used, to upload all the files? Step by step.

One by one from the folder and then select upload.

G-Redos, 2024-09-20:
Use this method:
**https://www.youtube.com/watch?v=9i1nDUoDRcI**

G-Redos, 2024-09-21:
You then should see tabs, in your Arduino IDE, with all the files open.

closed 2024-09-21 by Tonka7aBG