A-Emile/Lime_Gen3_IoT_Replacement #23: Code Problems

G-Redos, 2024-09-20:
Hi guys,

I am fairly new to Arduino.
I have managed to figure out majority of the stuff myself, but I am stuck on the "uploading code" to the esp32 part.

I understand that this is a functioning code base and works with the ESP32?
If it is functioning, what CRC versions are you using.
I have the newest one installed on the 1.8 IDE, and I am getting a lot of errors.
Do I need to use a specific version?

Also do I need to use a specific verison for the BLE stack?

I have posted the errors below.

Am I doing something wrong?

Arduino Sketchbook\LimeIoT\controller.ino: In function 'void readController()':

Arduino Sketchbook\LimeIoT\controller.ino:19:36: warning: 'uint16_t crc16(const uint8_t*, crc_size_t, uint16_t, uint16_t, uint16_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC16() instead [-Wdeprecated-declarations]
   19 |       uint16_t new_checksum = crc16(command, sizeof(command) - 2, 0x1021, 0x0000, 0x0000, false, false);
      |                               ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Arduino Sketchbook\LimeIoT\LimeIoT.ino:2:
Arduino Sketchbook\libraries\CRC\src/CRC.h:87:10: note: declared here
   87 | uint16_t crc16(
      |          ^~~~~

Arduino Sketchbook\LimeIoT\controller.ino:34:12: warning: unused variable 't' [-Wunused-variable]
   34 |       char t = Serial.read();
      |            ^

Arduino Sketchbook\LimeIoT\display.ino: In function 'void turnOffDisplayLed()':

Arduino Sketchbook\LimeIoT\display.ino:17:23: warning: 'uint8_t crc8(const uint8_t*, crc_size_t, uint8_t, uint8_t, uint8_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC8() instead [-Wdeprecated-declarations]
   17 |   byte checksum = crc8(myByte, sizeof(myByte), crc_poly, crc_init, crc_xorout, crc_refin, crc_refout);
      |                   ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Arduino Sketchbook\libraries\CRC\src/CRC.h:67:9: note: declared here
   67 | uint8_t crc8(
      |         ^~~~

Arduino Sketchbook\LimeIoT\display.ino: In function 'void sendDisplayCommand(int, byte, String)':

Arduino Sketchbook\LimeIoT\display.ino:50:27: warning: 'uint8_t crc8(const uint8_t*, crc_size_t, uint8_t, uint8_t, uint8_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC8() instead [-Wdeprecated-declarations]
   50 |   uint8_t crc_value = crc8(input_bytes, input_len / 2, crc_poly, crc_init, crc_xorout, crc_refin, crc_refout);
      |                       ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Arduino Sketchbook\libraries\CRC\src/CRC.h:67:9: note: declared here
   67 | uint8_t crc8(
      |         ^~~~

Arduino Sketchbook\LimeIoT\loop.ino: In function 'void loop()':

Arduino Sketchbook\LimeIoT\loop.ino:24:48: warning: narrowing conversion of 'speed' from 'float' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   24 |   byte txByte[] = { isUnlocked, unlockForEver, speed, battery, throttle, lightIsOn, controllerIsOn, isCharging, alarmIsOn };
      |                                                ^~~~~

Arduino Sketchbook\LimeIoT\loop.ino:25:27: warning: narrowing conversion of 'max_speed' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   25 |   byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                           ^~~~~~~~~

Arduino Sketchbook\LimeIoT\loop.ino:25:38: warning: narrowing conversion of 'alarm_delay' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   25 |   byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                                      ^~~~~~~~~~~

Arduino Sketchbook\LimeIoT\loop.ino:25:51: warning: narrowing conversion of 'alarm_freq' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   25 |   byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                                                   ^~~~~~~~~~

Arduino Sketchbook\LimeIoT\loop.ino:25:63: warning: narrowing conversion of 'alarm_reps' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   25 |   byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                                                               ^~~~~~~~~~

Arduino Sketchbook\LimeIoT\settings.ino: In member function 'virtual void SettingsBLECallback::onWrite(BLECharacteristic*)':

Arduino Sketchbook\LimeIoT\settings.ino:33:29: warning: narrowing conversion of 'max_speed' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   33 |     byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                             ^~~~~~~~~

Arduino Sketchbook\LimeIoT\settings.ino:33:40: warning: narrowing conversion of 'alarm_delay' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   33 |     byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                                        ^~~~~~~~~~~

Arduino Sketchbook\LimeIoT\settings.ino:33:53: warning: narrowing conversion of 'alarm_freq' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   33 |     byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                                                     ^~~~~~~~~~

Arduino Sketchbook\LimeIoT\settings.ino:33:65: warning: narrowing conversion of 'alarm_reps' from 'int' to 'byte' {aka 'unsigned char'} [-Wnarrowing]
   33 |     byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
      |                                                                 ^~~~~~~~~~

Arduino Sketchbook\LimeIoT\setup.ino: In function 'void setup()':

Arduino Sketchbook\LimeIoT\setup.ino:71:11: warning: unused variable 'oob_support' [-Wunused-variable]
   71 |   uint8_t oob_support = ESP_BLE_OOB_DISABLE;
      |           ^~~~~~~~~~~

Arduino Sketchbook\LimeIoT\utilsC.ino: In function 'void lockScooter()':
utilsC:4:16: error: 'LOCK_RELAY' was not declared in this scope; did you mean 'LOCK_EX'?
    4 |   digitalWrite(LOCK_RELAY, HIGH);
      |                ^~~~~~~~~~
      |                LOCK_EX

Arduino Sketchbook\LimeIoT\utilsC.ino: In function 'void unlockScooter()':
utilsC:11:16: error: 'LOCK_RELAY' was not declared in this scope; did you mean 'LOCK_EX'?
   11 |   digitalWrite(LOCK_RELAY, LOW);
      |                ^~~~~~~~~~
      |                LOCK_EX

Arduino Sketchbook\LimeIoT\utilsO.ino: At global scope:
utilsO:1:6: error: redefinition of 'void lockScooter()'
    1 | void lockScooter() {
      |      ^~~~~~~~~~~

Arduino Sketchbook\LimeIoT\utilsC.ino:2:6: note: 'void lockScooter()' previously defined here
    2 | void lockScooter() {
      |      ^~~~~~~~~~~

utilsO:6:6: error: redefinition of 'void unlockScooter()'
    6 | void unlockScooter() {
      |      ^~~~~~~~~~~~~

Arduino Sketchbook\LimeIoT\utilsC.ino:9:6: note: 'void unlockScooter()' previously defined here
    9 | void unlockScooter() {
      |      ^~~~~~~~~~~~~

Using library CRC at version 1.0.3 in folder: Arduino Sketchbook\libraries\CRC 

Using library BLE at version 3.0.4 in folder: ArduinoData\packages\esp32\hardware\esp32\3.0.4\libraries\BLE 
exit status 1

'LOCK_RELAY' was not declared in this scope; did you mean 'LOCK_EX'?

aIecxs, 2024-09-21:
Arduino 1.8.19 should work for Espressif v2.0.13. the CRC usage has changed:

https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/pull/12/commits/54fffa153b374851289b7b039c9ce6b4596bb582#diff-e7316332106dc654d8f30af0cdf756e0c4567aba2bc82414f8e3a00ef53658a4

G-Redos, 2024-09-21:
I am using Arduino 1.8.19 (windows store 1.8.57.0 build)
ESP32 version: ESP32-WROOM-32D DEVKIT V4

Should I install a lower version of the CRC library?

G-Redos, 2024-09-21:
> Arduino 1.8.19 should work for Espressif v2.0.13. the CRC usage has changed:
> 
> [54fffa1#diff-e7316332106dc654d8f30af0cdf756e0c4567aba2bc82414f8e3a00ef53658a4](https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/commit/54fffa153b374851289b7b039c9ce6b4596bb582#diff-e7316332106dc654d8f30af0cdf756e0c4567aba2bc82414f8e3a00ef53658a4)

So to my understanding, I should switch to CRC v1.0.2?
Should I also use your code base instead of the LimeIoT file?

G-Redos, 2024-09-26:
Does anyone know what is the correct procedure, in uploading the code to the ESP32?
Should I create a new sketch and do "ESP32 sketch data upload" option?
or
Open all the files in the IDE and then press upload?

I have used alecxs's updated code in the process, and I am getting the following error:

exit status 1
variable or field 'sendDisplayLED' declared void

aIecxs, 2024-09-26:
'sendDisplayLED' is declared in 'display.h' which is not up-to-date in this branch. delete everything and start with the 'themed' branch. you need to "ESP32 Sketch Data Upload" the mp3 files only once.

@A-Emile hasn't created a 'themed' branch yet, so while this pull request is pending you can clone from this fork instead.
```
git clone -b themed https://github.com/aIecxs/Lime_Gen3_IoT_Replacement.git
```

G-Redos, 2024-09-26:
> 'sendDisplayLED' is declared in 'display.h' which is not up-to-date in this branch. delete everything and start with the 'themed' branch. you need to "ESP32 Sketch Data Upload" the mp3 files only [...]

Hi alecxs,

Thank you for the reply.
I have previously managed to locate your repo and download all the files.
I have done the ESP32 Sketch Data Upload for the mp3 files.
I have changed type std::string to String in line 3, and in the rest of the .ino files.
I have installed CRC v1.0.2, for compatibility.

Using library CRC at version 1.0.2 
Using library BLE at version 3.0.5 
Using library ESP8266Audio at version 1.9.9 
Using library LittleFS at version 3.0.5 
Using library FS at version 3.0.5
Using library SPIFFS at version 3.0.5 
Using library HTTPClient at version 3.0.5 
Using library Network at version 3.0.5
Using library NetworkClientSecure at version 3.0.5 
Using library SD at version 3.0.5 
Using library SPI at version 3.0.5 

Using Arduino 1.8.19, I am still receiving error output:

In the settings.ino file, line 28:
  sendControllerCommand(data, value.size());

This error appears: 
exit status 1
'class String' has no member named 'size'

aIecxs, 2024-09-26:
> I have changed type std::string to String in line 3, and in the rest of the .ino files.

Why?

> This error appears: exit status 1 'class String' has no member named 'size'

it's because of the changes you made

G-Redos, 2024-09-26:
> > I have changed type std::string to String in line 3, and in the rest of the .ino files.
> 
> Why?
> 
> > This error appears: exit status 1 'class String' has no member named 'size'
> 
> it's because of the changes you made

Ok, I have went back into the files and made the change back to the original files in your repo.|

But now I have getting this:

Exception 0x8b04488b 0xccccccd5 0x8b64cccc 0x2000bfff
PC=0x2000bfff

runtime: unknown pc 0x2000bfff
stack: frame={sp:0xf76e2000, fp:0x0} stack=[0x12856000,0x12858000)

runtime: unknown pc 0x2000bfff
stack: frame={sp:0xf76e2000, fp:0x0} stack=[0x12856000,0x12858000)

eax     0xcf130000
ebx     0x98003ffd
ecx     0x3ffeeffb
edx     0x9070ab38
edi     0x99999800
esi     0x99999999
ebp     0xd41fd4d9
esp     0xf76e2000
eip     0x2000bfff
eflags  0x4009b70d
cs      0x9f43f76e
fs      0x7800ffff
gs      0x0
ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder returned 2
Error compiling for board ESP32 Dev Module.

Is it my settings or am I missing something?

G-Redos, 2024-09-26:
exit status 1
'class String' has no member named 'size'

File: settings.ino
Line 28: sendControllerCommand(data, value.size());

G-Redos, 2024-09-26:
> > I have changed type std::string to String in line 3, and in the rest of the .ino files.
> 
> Why?
> 
> > This error appears: exit status 1 'class String' has no member named 'size'
> 
> it's because of the changes you made

You said to do it here:
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/issues/16

G-Redos, 2024-09-26:
text section exceeds available space in board

Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

Error compiling for board ESP32 Dev Module.

aIecxs, 2024-09-26:
> You said to do it here:
> #16

it's a long time ago I compiled the code. I have Arduino 1.8.19 installed under linux.

these are the Additional Boards Manager URLs:

```
https://espressif.github.io/arduino-esp32/package_esp32_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://www.pjrc.com/teensy/package_teensy_index.json
http://dan.drown.org/stm32duino/package_STM32duino_index.json
```

this is what I see under Boards Manager:

```
Arduino AVR Boards Built-In by Arduino version 1.8.3 INSTALLED
esp32 by Espressif Systems version 2.0.9 INSTALLED
STM32F1xx/GD32F1xx boards by stm32duino version 2022.9.26 INSTALLED
```

this is what I see under Library Manager:

```
Bridge Built-In by Arduino Version 1.7.0 INSTALLED
Esplora Built-In by Arduino Version 1.0.4 INSTALLED
Ethernet Built-In by Arduino Version 2.0.0 INSTALLED
Firmata Built-In by Firmata team Version 2.5.8 INSTALLED
GSM Built-In by Arduino Version 1.0.6 INSTALLED
Keyboard Built-In by Arduino Version 1.0.2 INSTALLED
LiquidCrystal Built-In by Arduino Version 1.0.7 INSTALLED
Mouse Built-In by Arduino Version 1.0.1 INSTALLED
Robot Control Built-In by Arduino Version 1.0.4 INSTALLED
Robot IR Remote Built-In by Arduino Version 2.0.0 INSTALLED
Robot Motor Built-In by Arduino Version 1.0.3 INSTALLED
SD Built-In by Arduino Version 1.2.4 INSTALLED
Servo Built-In by Arduino Version 1.1.8 INSTALLED
SpacebrewYun Built-In by Julio Terra Version 1.0.2 INSTALLED
Stepper Built-In by Arduino Version 1.1.3 INSTALLED
Temboo Built-In by Temboo Version 1.2.1 INSTALLED
TFT Built-In by Arduino Version 1.0.6 INSTALLED
WiFi Built-In by Arduino Version 1.2.7 INSTALLED
Adafruit Circuit Playground Built-In by Adafruit Version 1.11.3 INSTALLED
CRC by Rob Tillaart Version 1.0.2 INSTALLED
ESP8266Audio by Earle F. Philhower, III Version 1.9.7 INSTALLED
FreeRTOS by Phillip Stevens Version 10.5.1-0 INSTALLED
TaskScheduler by Anatoli Arkhipenko Version 3.7.0 INSTALLED
```

this is the Board I selected:

```
ESP32 Arduino -> DOIT ESP32 DEVKIT V1
```

not sure where you got the version of the libraries from, I have none of the libs you listed?

Downgrade esp32 by Espressif Systems version 2.0.13 to avoid exceeding of program storage space. Maximum is 1310720 bytes.

G-Redos, 2024-09-26:
> > You said to do it here:
> > #16
> 
> this is another issue regarding Arduino 2.3.2, not sure why you linked that here.
> 
> it's a long time ago I compiled the code. I have Arduino 1.8.19 installed under linux.
> 
> these are the Additional Boards Manager URLs:
> 
> ```
> https://espressif.github.io/arduino-esp32/package_esp32_index.json
> https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
> https://www.pjrc.com/teensy/package_teensy_index.json
> http://dan.drown.org/stm32duino/package_STM32duino_index.json
> ```
> 
> this is what I see under Boards Manager:
> 
> ```
> Arduino AVR Boards Built-In by Arduino version 1.8.3 INSTALLED
> esp32 by Espressif Systems version 2.0.9 INSTALLED
> STM32F1xx/GD32F1xx boards by stm32duino version 2022.9.26 INSTALLED
> ```
> 
> this is what I see under Library Manager:
> 
> ```
> Bridge Built-In by Arduino Version 1.7.0 INSTALLED
> Esplora Built-In by Arduino Version 1.0.4 INSTALLED
> Ethernet Built-In by Arduino Version 2.0.0 INSTALLED
> Firmata Built-In by Firmata team Version 2.5.8 INSTALLED
> GSM Built-In by Arduino Version 1.0.6 INSTALLED
> Keyboard Built-In by Arduino Version 1.0.2 INSTALLED
> LiquidCrystal Built-In by Arduino Version 1.0.7 INSTALLED
> Mouse Built-In by Arduino Version 1.0.1 INSTALLED
> Robot Control Built-In by Arduino Version 1.0.4 INSTALLED
> Robot IR Remote Built-In by Arduino Version 2.0.0 INSTALLED
> Robot Motor Built-In by Arduino Version 1.0.3 INSTALLED
> SD Built-In by Arduino Version 1.2.4 INSTALLED
> Servo Built-In by Arduino Version 1.1.8 INSTALLED
> SpacebrewYun Built-In by Julio Terra Version 1.0.2 INSTALLED
> Stepper Built-In by Arduino Version 1.1.3 INSTALLED
> Temboo Built-In by Temboo Version 1.2.1 INSTALLED
> TFT Built-In by Arduino Version 1.0.6 INSTALLED
> WiFi Built-In by Arduino Version 1.2.7 INSTALLED
> Adafruit Circuit Playground Built-In by Adafruit Version 1.11.3 INSTALLED
> CRC by Rob Tillaart Version 1.0.2 INSTALLED
> ESP8266Audio by Earle F. Philhower, III Version 1.9.7 INSTALLED
> FreeRTOS by Phillip Stevens Version 10.5.1-0 INSTALLED
> TaskScheduler by Anatoli Arkhipenko Version 3.7.0 INSTALLED
> ```
> 
> this is the Board I selected:
> 
> ```
> ESP32 Arduino -> DOIT ESP32 DEVKIT V1
> ```
> 
> not sure where you got the version of the libraries from, I have none of the libs you listed?
> 
> Downgrade esp32 by Espressif Systems version 2.0.13 to avoid exceeding of program storage space. Maximum is 1310720 bytes.

Hi Alecxs,

I have linked issue #16 because I get the exact same problem using the IDE Arduino 1.8.19.

The libs I have listed, I downloaded from the Library Manager/Type/Arduino and Recommended. 

**Configuration:**
IDE: Arduino 1.8.19
[... 46 lines cut ...]
**2. REPO: https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/tree/main/LimeIoT** 
-with the newest branch merge in files commands.ino and settings.ino

When compiling the code, with Board/DOIT ESP32 DEVKIT V1, the following error is provided:
exit status 1
conversion from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to non-scalar type 'String' requested
FILE: settings.ino
LINE 4: String preValue = pCharacteristic->getValue(); // remove str::string (issues/16)
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/blob/main/LimeIoT/settings.ino

When compiling the code, with Board/ESP32-WROOM-DA Module, the following error is provided: 
exit status 1
conversion from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to non-scalar type 'String' requested
FILE: settings.ino
LINE 4: String preValue = pCharacteristic->getValue(); // remove str::string (issues/16)
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/blob/main/LimeIoT/settings.ino

When compiling the code, with Board/ESP32 Dev Module, the following error is provided: 
exit status 1
conversion from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to non-scalar type 'String' requested
FILE: settings.ino
LINE 4: String preValue = pCharacteristic->getValue(); // remove str::string (issues/16)
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/blob/main/LimeIoT/settings.ino

All day today, with no success in flashing the ESP32, using 2 repos.

aIecxs, 2024-09-26:
You're using Arduino ESP32 Boards - version 2.0.17
you should use esp32 by Espressif Systems - version 2.0.13

G-Redos, 2024-09-26:
[image]

aIecxs, 2024-09-26:
please note the difference

> **Boards Manager:**
> Arduino AVR Boards - version 1.8.6
> Arduino ESP32 Boards - version 2.0.17

G-Redos, 2024-09-26:
> please note the difference
> 
> > **Boards Manager:**
> > Arduino AVR Boards - version 1.8.6
> > Arduino ESP32 Boards - version 2.0.17

This is what I exactly have installed on the IDE:

[image]

aIecxs, 2024-09-26:
uninstall Arduino ESP32 Boards

G-Redos, 2024-09-26:
[image]

aIecxs, 2024-09-26:
I have tested the themes branch, it compiles fine under Windows and Linux.

G-Redos, 2024-09-26:
ESP32 Dev Module error:
exit status 1
format '%d' expects argument of type 'int', but argument 4 has type 'long unsigned int' [-Werror=format=]

DOIT ESP32 error:
exit status 1
format '%d' expects argument of type 'int', but argument 4 has type 'long unsigned int' [-Werror=format=]

Maybe I should uninstall the IDE, and reinstall it again?

aIecxs, 2024-09-26:
delete the messed up repo and compile the themed branch
```
git clone -b themed https://github.com/aIecxs/Lime_Gen3_IoT_Replacement.git
```

G-Redos, 2024-09-26:
1. Uninstalled Ardiuno IDE from MS store.
2. Reinstalled Ardiuno IDE from MS store version 1.8.19
3. Updated additional boards manager URLs
4. Installed esp32 by Espressif Systems version 2.0.13
5. Installed CRC lib version 1.0.3
6. Installed ESP8266Audio lib version 1.9.9
7. Downloaded esp32fs.jar https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/tag/1.1
- https://github.com/aIecxs/Lime_Gen3_IoT_Replacement/blob/themed/LimeIoT/I2Saudio.ino
- LINE 10 needs updating with the new link
10. Unpacked the tool into tools directory (the path will look like <home_dir>/Arduino/tools/ESP32FS/tool/esp32fs.jar).
11. Restarted IDE
12. Downloaded the following repo: https://github.com/aIecxs/Lime_Gen3_IoT_Replacement.git
13. Unpacked with 7-zip entire repo
14. Copied LimeIoT Folder to Arduino Sketches folder
15. Opened limeiot.ino file and loaded all the remaining files
16. Set board to ESP32 Dev Module
17. Set Baud rate to 115200
18. Set COM Port
19. Verified connection to ESP32 using serial monitor set at Baud rate 115200
20. Clicked upload

**RESULTS:** 
Wrote 1283824 bytes (797489 compressed) at 0x00010000 in 70.7 seconds (effective 145.2 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...

21. Clicked ESP32 Sketch Data upload

**RESULTS:**
Wrote 1441792 bytes (101506 compressed) at 0x00290000 in 13.6 seconds (effective 850.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

G-Redos, 2024-09-26:
Thank you @aIecxs !

But still, the repos need to be overhauled and organized better.
Currently there are 2 branches.
It makes everything confusing for someone trying to get it done.

At least anyone new, trying to replicate the process, will have a full guideline on this repo.

The code base should be sorted into:

**Arduino IDE version 1.8:**
-With speaker support
-Without speaker support 
-With sensor support
-Without sensor support

**Arduino IDE version 2.0:**
-With speaker support
-Without speaker support 
-With sensor support
-Without sensor support

aIecxs, 2024-09-27:
someone changed 'main' branch to work with Arduino 2.0 and forgot to update README.md.

careful read README.md in 'themed' branch there is outlined the Arduino 1.8 requirements with link to pcb board.

for support ask in forum.

G-Redos, 2024-09-27:
Hi @aIecxs ,

After connecting the IOT to the main device, I am not able to see the ESP32 Bluetooth connection.
No transmission is being picked up by the NRF application on android.

FILE: LimeIoT.ino 
has the UUID number, but I am not picking it up.

// BLE
#define **SERVICE_UUID** "653bb0e0-1d85-46b0-9742-3b408f4cb83f"
#define CHARACTERISTIC_UUID_MAIN "00c1acd4-f35b-4b5f-868d-36e5668d0929"
#define CHARACTERISTIC_UUID_SETTINGS "7299b19e-7655-4c98-8cf1-69af4a65e982"
#define CHARACTERISTIC_UUID_DEBUG "83ea7700-6ad7-4918-b1df-61031f95cf62"

Do I need to reset the ESP32, to startup the Bluetooth function?
Or is there another process?

G-Redos, 2024-09-27:
17:48:29.249 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:48:29.249 -> configsip: 0, SPIWP:0xee
17:48:29.249 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:48:29.249 -> mode:DIO, clock div:1
17:48:29.249 -> load:0x3fff0030,len:1344
17:48:29.249 -> load:0x40078000,len:13964
17:48:29.249 -> load:0x40080400,len:3600
17:48:29.249 -> entry 0x400805f0
17:48:29.673 -> E (13) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1225:error: Corruptedets Jul 29 2019 12:21:46

G-Redos, 2024-09-27:
18:03:23.977 -> E (11) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1225:error: Corrupted

aIecxs, 2024-09-27:
I am afraid your ESP32 is not compatible with 'themed' branch. Try again with http://github.com/lorol/arduino-esp32fs-plugin/releases

G-Redos, 2024-09-27:
You cannot also Auto Format the code in the Arduino IDE 1.8, if you do that, you won't be able to flash the esp32.
You will be forced to start over with a new code base.

The current error I am getting, is an acceptable error, and I am able to connect via Bluetooth now.

I suggest adding the comment in the LimeIoT.ino file, that you can change the name of the scooter, and the password.

aIecxs, 2024-09-27:
it's not acceptable error because the ESP32 will crash when trying to play mp3 / file not found. not sure what you mean with Auto Format but you can repartition and reformat the LittleFS maybe that solves the error. if not check the boards.txt for correct flash size / partitioning

G-Redos, 2024-09-27:
> it's not acceptable error because the ESP32 will crash when trying to play mp3 / file not found. not sure what you mean with Auto Format but you can repartition and reformat the LittleFS maybe that solves the error. if not check the boards.txt for correct flash size / partitioning

In the Arduino IDE 1.8, you have the Ctrl + T shortcut to auto format the code.

G-Redos, 2024-09-28:
->esptool.py v4.5.1
->Serial port COM5
->Connecting....
->Chip is ESP32-D0WD-V3 (revision v3.1)
->Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
->Crystal is 40MHz
->MAC: c8:2e:18:8d:e2:3c
->Uploading stub...
->Running stub...
->Stub running...
->Erasing flash (this may take a while)...
->Chip erase completed successfully in 6.6s
->Hard resetting via RTS pin...

G-Redos, 2024-09-28:
Upload of sketch data is complete.

->esptool.py v4.5.1
->Serial port COM5
->Connecting.....
->Chip is ESP32-D0WD-V3 (revision v3.1)
->Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
->Crystal is 40MHz
->MAC: c8:2e:18:8d:e2:3c
->Uploading stub...
->Running stub...
->Stub running...
->Configuring flash size...
->Auto-detected Flash size: 4MB
->Flash will be erased from 0x00290000 to 0x003effff...
->Compressed 1441792 bytes to 95694...
->Writing at 0x00290000... (16 %)
->Writing at 0x003577df... (33 %)
->Writing at 0x0035b988... (50 %)
->Writing at 0x0035fb80... (66 %)
->Writing at 0x003648eb... (83 %)
->Writing at 0x00369f2c... (100 %)
->Wrote 1441792 bytes (95694 compressed) at 0x00290000 in 10.4 seconds (effective 1112.5 kbit/s)...
->Hash of data verified.
->
->Leaving...
->Hard resetting via RTS pin...

closed 2024-10-03 by G-Redos