A-Emile/Lime_Gen3_IoT_Replacement #24: "Flash the controller with unlocked firmware."

gc9n, 2024-10-12:
hi , 
sorry if been replied before but , 
what exactly does the firmare that unlocks the controller with the ST link , ?  what exactly unlocks? speed ? 

WITH only this firmare update adn WITHOUT the Esp32 that sends commands , the scooter is a brick?

G-Redos, 2024-10-14:
No, you can purchase online a different controller and it will work also.

closed 2024-10-15 by gc9n

reuzurq, 2024-10-19:
**@G-Redos dont comment on things you have absolutely no understanding of.**

@gc9n:
"Unlocked" firmware is an old firmware used before Lime in 2019 patched the ESC unlocking command.
Does not affect speed or any other variables, just uses old commands that are already public.

Without flashing that firmware, you will be able to control every scooter aspect (like lights, speed, etc.), **but you wont be able to unlock and ride it**. You can use boards other than ESP32, like Arduino (with your own code) to control the scooter.

G-Redos, 2025-02-09:
@reuzurq I think you need to do a little more research online, and see other projects with aftermarket parts sourced from other suppliers online. 
If you want, I can leave links on here to other projects, where the esp32 was not used and a different motor controller was installed.
Just let me know.