A-Emile/Lime_Gen3_IoT_Replacement #25: Pin numbers

gc9n, 2024-10-15:
What exactly pin #1 into this pin headers? 
I will desolder and use this one.
[image]

G-Redos, 2024-10-15:
The cables have colors.
Each cable has a connection to a specific GPIO.

gc9n, 2024-10-16:
Thanks for your answer , 
So i follow the colours of this diagram to the actual connector 
i see on the cable harness , Pin 3 and Pin 4 are Red coloured , ok i will use a multimeter.

[image]

aIecxs, 2024-10-16:
the numbering of the connector is not correct. you can look into datasheet [here](https://www.tme.eu/en/details/df13ea20dp1.25v65/raster-signal-connectors-1-25mm/hirose/df13ea-20dp-1-25v-65) the pin numbering is 180° turned.

gc9n, 2024-10-16:
Let me ask again maybe I didn't asked the question correctly 

I want to desolder this and solder ON it the ESP32 pins.
However I don't know the numbering .
I will take a look on the datasheet you send me thanks 

[image]

G-Redos, 2024-10-16:
Look at your ESP32.
They have specific GPIO pins.
The cables are color coded.
You have two cables, one from the main controller, and the other from the display.
The color coded cables, go specifically to the designated GPIO pin to the ESP32.

Yes it is possible, to do what you want.
But you will have to match the cable layout to the GPIO pin layout of the ESP32.

aIecxs, 2024-10-17:
Yes, I had already understood that. I said the connector is rotated by 180° degree. See the triangle marking on the connector, Pin 1 = Display TX.
[image]

Here is the full pinout:
https://forum.xda-developers.com/posts/89114905

gc9n, 2024-10-17:
thanks guys

gc9n, 2024-10-17:
[image]
I am 1% before I finish 
Can you guys explain controller lock goes to GPIO12?
And COntroller charge to pin 2??

gc9n, 2024-10-17:
[image]
Everything is Flawless 
The only problem is that the battery shows 0%

gc9n, 2024-10-17:
I doubled checked everything 
 
Throttle working
Light working 
Battery is 0% but is 100%
And km per h are. 0 which is false.

Ps the controller is flashed with st link from the known place

gc9n, 2024-10-17:
the controller is flashed with 
[https://www.dropbox.com/scl/fo/v255vxco67f128a9j56qc/ADV5v-_SYfOrOjPAkJEbjjs/LIME%20File/HEX%20Backup?dl=0&rlkey=q2jcr5qzxdxrdi34t774icoog&subfolder_nav_tracking=1
](LINK )
the Unlocked 25k

gc9n, 2024-10-17:
long story short everything is working , even the brake light , motor is working, 
i just get no info for Speed and Battery, i am chinging the code that i use

https://github.com/aIecxs/Lime_Gen3_IoT_Replacement/blob/2.3.x/LimeIoT/LimeIoT.ino

aIecxs, 2024-10-17:
when you connect the charger does it display battery charging?

gc9n, 2024-10-17:
> when you connect the charger does it display battery charging?

Even though pin2 is connected to GPIO33  doesnt give me the status.  
So I assume that 
something is wrong on the readController()  , apparently the incoming bytes are not read correctly , 
i already update the "silver "  controller  to locked 25  , unlocked 25 with the same results.
theres is no communication with the serial on the controller , 
i will double check if i messed the TX RX pins

PS  My Sketch is the 2.3.x

gc9n, 2024-10-17:
No , its code related the TX RX ESP32 <-> RX TX CONtroller   is fine ,
and also the GPIO33 is not related to the UART communication

gc9n, 2024-10-17:
The light Command is working just fine.... very weird.   so there is communication with the Controller and the ESP however some incoming value are not recognized?

aIecxs, 2024-10-17:
When charger is connected, the controller is powered on automatically and is continuously sending data, regardless of the lock state. Use any USB to UART Bridge (CP2102 or CH340 or something) to verify the data is sent/received on GPIO 1 + 3.

The front light is controlled by the display controller on GPIO 16 + 17.

gc9n, 2024-10-17:
What CRC library do i have to use?  i am using 1.0.3 Rob tillard

gc9n, 2024-10-17:
Ok i tried my module , on another set of Controller/monitor and is working , battery and speed, 
So i guess is cable related, i dont know wich on have to find out

G-Redos, 2024-10-18:
What do you mean exactly, you connected your setup with the ESP32 to another device and the results are working as they should?

gc9n, 2024-10-18:
i have no idea why , but it works now , i uninstall and reinstall all the cables , and is fine

closed 2024-10-18 by gc9n

G-Redos, 2024-10-18:
Cables connecting the IOT port to the ESP32, or the main cables from the motor controller to the ESP32?