A-Emile/Lime_Gen3_IoT_Replacement #26: Something is very wrong with the Code , the version of IDE ?

gc9n, 2024-10-15:
Hi i used latest Arduino IDE 2.3.x and i got lots of errors.   after some reading here on the issues , i used Arduino 1.8.19   my board is Wroom ESP32   CRC by Rob Tillaart Version 1.0.2 INSTALLED
, what execatly is happening with the versions?  
i have a banch of error when i compile .

Does anyone have a VM with this Working to share ?

In file included from C:\Users\Gchat\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEAdvertising.h:15,
                 from C:\Users\Gchat\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEServer.h:19,
                 from C:\Users\Gchat\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDevice.h:18,
                 from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:3:
C:\Users\Gchat\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\commands.ino: In member function 'virtual void MainBLECallback::onWrite(BLECharacteristic*)':
commands:3:48: error: conversion from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to non-scalar type 'String' requested
     String preValue = pCharacteristic->getValue(); // remove str::string (issues/16)
                       ~~~~~~~~~~~~~~~~~~~~~~~~~^~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\controller.ino: In function 'void readController()':
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\controller.ino:19:103: warning: 'uint16_t crc16(const uint8_t*, crc_size_t, uint16_t, uint16_t, uint16_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC16() instead [-Wdeprecated-declarations]
       uint16_t new_checksum = crc16(command, sizeof(command) - 2, 0x1021, 0x0000, 0x0000, false, false);
                                                                                                       ^
In file included from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:2:
C:\Users\Gchat\Documents\Arduino\libraries\CRC\src/CRC.h:87:10: note: declared here
 uint16_t crc16(
          ^~~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\controller.ino:19:103: warning: 'uint16_t crc16(const uint8_t*, crc_size_t, uint16_t, uint16_t, uint16_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC16() instead [-Wdeprecated-declarations]
       uint16_t new_checksum = crc16(command, sizeof(command) - 2, 0x1021, 0x0000, 0x0000, false, false);
                                                                                                       ^
In file included from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:2:
C:\Users\Gchat\Documents\Arduino\libraries\CRC\src/CRC.h:87:10: note: declared here
 uint16_t crc16(
          ^~~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\display.ino: In function 'void turnOffDisplayLed()':
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\display.ino:17:101: warning: 'uint8_t crc8(const uint8_t*, crc_size_t, uint8_t, uint8_t, uint8_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC8() instead [-Wdeprecated-declarations]
   byte checksum = crc8(myByte, sizeof(myByte), crc_poly, crc_init, crc_xorout, crc_refin, crc_refout);
                                                                                                     ^
In file included from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:2:
C:\Users\Gchat\Documents\Arduino\libraries\CRC\src/CRC.h:67:9: note: declared here
 uint8_t crc8(
         ^~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\display.ino:17:101: warning: 'uint8_t crc8(const uint8_t*, crc_size_t, uint8_t, uint8_t, uint8_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC8() instead [-Wdeprecated-declarations]
   byte checksum = crc8(myByte, sizeof(myByte), crc_poly, crc_init, crc_xorout, crc_refin, crc_refout);
                                                                                                     ^
In file included from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:2:
C:\Users\Gchat\Documents\Arduino\libraries\CRC\src/CRC.h:67:9: note: declared here
 uint8_t crc8(
         ^~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\display.ino: In function 'void sendDisplayCommand(int, byte, String)':
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\display.ino:50:109: warning: 'uint8_t crc8(const uint8_t*, crc_size_t, uint8_t, uint8_t, uint8_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC8() instead [-Wdeprecated-declarations]
   uint8_t crc_value = crc8(input_bytes, input_len / 2, crc_poly, crc_init, crc_xorout, crc_refin, crc_refout);
                                                                                                             ^
In file included from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:2:
C:\Users\Gchat\Documents\Arduino\libraries\CRC\src/CRC.h:67:9: note: declared here
 uint8_t crc8(
         ^~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\display.ino:50:109: warning: 'uint8_t crc8(const uint8_t*, crc_size_t, uint8_t, uint8_t, uint8_t, bool, bool, crc_size_t)' is deprecated: Use calcCRC8() instead [-Wdeprecated-declarations]
   uint8_t crc_value = crc8(input_bytes, input_len / 2, crc_poly, crc_init, crc_xorout, crc_refin, crc_refout);
                                                                                                             ^
In file included from C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\LimeIoT.ino:2:
C:\Users\Gchat\Documents\Arduino\libraries\CRC\src/CRC.h:67:9: note: declared here
 uint8_t crc8(
         ^~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\loop.ino: In function 'void loop()':
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\loop.ino:24:123: warning: narrowing conversion of 'speed' from 'float' to 'byte' {aka 'unsigned char'} inside { } [-Wnarrowing]
   byte txByte[] = { isUnlocked, unlockForEver, speed, battery, throttle, lightIsOn, controllerIsOn, isCharging, alarmIsOn };
                                                                                                                           ^
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\loop.ino:25:74: warning: narrowing conversion of 'max_speed' from 'int' to 'byte' {aka 'unsigned char'} inside { } [-Wnarrowing]
   byte settingsByte[] = { max_speed, alarm_delay, alarm_freq, alarm_reps };
                                                                          ^
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\loop.ino:25:74: warning: narrowing conversion of 'alarm_delay' from 'int' to 'byte' {aka 'unsigned char'} inside { } [-Wnarrowing]
[... 9 lines cut ...]
settings:7:24: error: 'class String' has no member named 'size'
     uint8_t data[value.size()];
                        ^~~~
settings:8:12: error: 'data' was not declared in this scope
     memcpy(data, value.c_str(), value.size());
            ^~~~
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\settings.ino:8:12: note: suggested alternative: 'atan'
     memcpy(data, value.c_str(), value.size());
            ^~~~
            atan
settings:8:39: error: 'class String' has no member named 'size'
     memcpy(data, value.c_str(), value.size());
                                       ^~~~
settings:29:45: error: 'class String' has no member named 'size'
           sendControllerCommand(data, value.size());
                                             ^~~~
[... cut ...]
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\settings.ino:33:76: warning: narrowing conversion of 'alarm_freq' from 'int' to 'byte' {aka 'unsigned char'} inside { } [-Wnarrowing]
C:\Users\Gchat\Downloads\Lime_Gen3_IoT_Replacement-main\Lime_Gen3_IoT_Replacement-main\LimeIoT\settings.ino:33:76: warning: narrowing conversion of 'alarm_reps' from 'int' to 'byte' {aka 'unsigned char'} inside { } [-Wnarrowing]
Multiple libraries were found for "BLEDevice.h"
 Used: C:\Users\Gchat\Documents\Arduino\libraries\ESP32_BLE_Arduino
 Not used: C:\Users\Gchat\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14

aIecxs, 2024-10-16:
you can try to change board "ESP32 Dev Module" -> Partition Scheme: custom, and compile the [2.3.x](https://github.com/aIecxs/Lime_Gen3_IoT_Replacement/tree/2.3.x) branch.

G-Redos, 2024-10-16:
What about the CRC library version?

gc9n, 2024-10-17:
> you can try to change board "ESP32 Dev Module" -> Partition Scheme: custom, and compile the [2.3.x](https://github.com/aIecxs/Lime_Gen3_IoT_Replacement/tree/2.3.x) branch.

its a mess
Compilation error: AudioFileSourceLittleFS.h: No such file or directory

aIecxs, 2024-10-17:
kindly read the instructions for the [2 3.x](https://github.com/aIecxs/Lime_Gen3_IoT_Replacement/tree/2.3.x) (themed) branch

gc9n, 2024-10-17:
It worked , compiled and uploaded succesfull to the WRoom.

closed 2024-10-17 by gc9n