A-Emile/Lime_Gen3_IoT_Replacement #27: 2 3.x (themed) branch  And Bluetooth

gc9n, 2024-10-17:
hi ,  2 3.x (themed) branch  the Bluetooth  cannot be paired , i changed the password , but nothing, password variable doesnt even used anywhere

gc9n, 2024-10-17:
setup.ino line 111 theres a hardcoded pass 123456

closed 2024-10-17 by gc9n

aIecxs, 2024-10-17:
kindly read the instructions before open new issues. I have quoted the relevant section below:

> The default bluetooth password is `123456789`. You can change it in the [ble_security.ino](../../tree/HEAD/LimeIoT/ble_security.ino) file.

If you're asking especially for the themed branch, please also read the instructions of the relevant branch #26:

> If you forgot to upload mp3 files, the [themed](../../tree/themed) branch will crash the ESP32 when playing audio / file not found.

aIecxs, 2024-10-17:
Kindly read the instructions more carefully. For troubleshooting, please visit the [forum](https://scootertalk.org/forum/viewtopic.php?t=5474&start=180). You're more likely to get help there than by spamming the GitHub issues.

Regarding #28 let me emphasize once more what you might have missed:
[image]