A-Emile/Lime_Gen3_IoT_Replacement #29: Display OK, Battery ok but motor speed nok and still lock

Nheliteuf, 2024-11-24:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Stephane/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Stephane/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Stephane/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=FR style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>Hello,<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>I need
your help please.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>I use recommended
hardware with EPS32.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>I loaded
LimeIoTWithoutSleep project (found here :
https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/releases/tag/latest)<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>I got a
correct display control with my phone : locked, unlock, battery state of charge
but unfortunately, no speed displayed and controller seems to be locked.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>I tested
IOT on a second Lime Gen 3 and same problem.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>So I
checked signals using DSLogic Series USB-based Logic Analyzer
(https://www.dreamsourcelab.com/product/dslogic-series/) with DSView software.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNormal style='margin-bottom:0cm;line-height:normal'><span
lang=EN-US style='mso-ansi-language:EN-US'>I checked that Iot sends the correct
sequence:<o:p></o:p></span></p>

<p class=MsoNormal style='margin-bottom:0cm;line-height:normal'><span
lang=EN-US style='mso-ansi-language:EN-US'>4643110100084C494D4542494B45BE8A<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>A trame
is sent every 233ms.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>Then, the
motor controller replies every 457ms.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>The
following answers were received:<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

Id | Concat
-- | --
26664 | 46580CFF002211000040000041399542008B4437520061F180020072014B00018200000000E00000F35F
26706 | 46580CFF002211000040000041399442008B4437520061F180021072014B00018200000000E000005F3E
26748 | 46580CFF002211000040000041399542008B4437520061F180020072014B00018200000000E00000F35F
26790 | 46580CFF002211000048000041399542008B4437560061F180020072014B00018200000000E00000F35F
26832 | 46580CFF002211000040000041399442008B4437520061F180020872014B00018200000000E000005F3E
A-Emile Ex | 46580CFF0022110000400000413F604200FF4464520061F180000072015C01598200000000E000000AB3

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>A-Emile
: Byte n°20 = 0x64 so 100% remaining battery<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>Me : Byte
n°20 = 0x37 so 55% remaining battery.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>But Byte
n°9 always stays at 0 even if the wheel is turning.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>Do you
see something wrong or is the motor controller locked ?<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>I attached
so screen capture and DSView project.<o:p></o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNoSpacing><span lang=EN-US style='mso-ansi-language:EN-US'>Thank
you for your help.<o:p></o:p></span></p>

</body>

</html>

IOT sends correct sequence to controller motor :
[image]

Then the motor controller answers :
[image]

Iot sends sends every 233ms :
[image]

Motor controller answers every 457ms :
[image]

Nheliteuf, 2024-11-24:
Here is the DSView free software project.

After a long motor controller ON time, i power up the IOT.

[controller first long puis esprssDSLogic PLus-la-241111-203427.zip](https://github.com/user-attachments/files/17893947/controller.first.long.puis.esprssDSLogic.PLus-la-241111-203427.zip)

A-Emile, 2024-11-24:
Have you flashed the unlocked firmware?

A-Emile, 2024-11-24:
Also note that the Arduino firmware is poorly written and maybe not working with current arduino versions. So you could try using an older arduino version to compile it.

Nheliteuf, 2024-12-01:
Yes I flashed controller with unlocked firmware.
Ok what version do you use for Arduino ?
Thank you.

john-leo93, 2024-12-19:
Did you figured out what it was? 

Regardless of seeing the IOT and the controller communicating ensure that you have a solid connection on TX RX and i noticed that on some boards the TX and RX are actually FLIPPED, indications printed on board do not match with the assigned pins. This can't be the case with you but worth mentioning for future problem solving.

Had a similar problem and it was the problem mentioned above, but mine was showing 0% battery and lights were malfunctioning, motor was working for 1 min etc.

Try Arduino IDE 1.8, in my case i couldn't get it to compile with 2+ at all.

Nheliteuf, 2025-01-05:
So, if I well understand, currently, there are 2 works in parallel:
A-emile team:
https://atollos.com/

Alexcs side:
https://scootertalk.org/forum/viewtopic.php?f=56&t=5474&start=180

With a friend, we will try both.

Nheliteuf, 2025-01-05:
[image]
[image]

Nheliteuf, 2025-01-05:
[image]