A-Emile/Lime_Gen3_IoT_Replacement #31: Disabling Bluetooth conpletely - Work standalone?

john-leo93, 2024-12-19:
Hello.

I'm having an issue with the Bluetooth connection on ESP32 which does not seem like it does always connect to two of phones i have, app works randomly you can press unlock and make it work without showing batt etc, other times it works okay.

Problem is yesterday i was almost stranded in bad weather due to this and was restarting the controller to make the scooter work for 1 min till it locks, then restart it again, did this till i got close to my house until i got sick and tried connecting to BT again, unlocked it, it disconnected but thankfully for unknown reasons the IOT kept working.

So anyone knows how to make the IOT always working when powered on without Bluetooth? 

Something simple like turn it on, communicate with the controller, lights always on etc.
What to exclude or include to the code?

This might be defeating its whole purpose but getting stranded due to phone being out of battery, lost, or Bluetooth acting weird, is nothing i want to play with.

I can send the always on command but it's not a solution unless i keep the scooter powered on indefinitely which is not possible.

For "safety" reasons i have a key switch enabling the 5v to ESP32 and it's enough, not different than any Xiaomi scooter etc which are turning on with a button.

If anyone can help with this i'll appreciate it.

reuzurq, 2025-01-13:
**the (poorly written) esp32 code already contains "unlock forever" function.**
Using apps like [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) you can send a `unlockforever` command to the IoT.

Valentin09hacker, 2025-02-26:
> **The (poorly written) esp32 code already contains "unlock forever" function.** Using apps like [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) you can send a `unlockforever` command to the IoT.

Because I'm trying to do that, and I am not familiar with the app Is there any way that you can provide us with a tutorial how to do that? Because I'm trying to and i cant do it

Valentin09hacker, 2025-02-26:
Can't we just use any other Ardino modules and just use a code that sends signal to the scooter controller to be unlocked forever?

aIecxs, 2025-05-06:
Just long press the unlock button in Limer app.

john-leo93, 2025-05-07:
Accidentally found a way to "glitch" it into working by its self unlocked forever without being connected to bluetooth.

1) I unlock it via bluetooth
(Btw i have issues with the ESP32 disconnecting-reconnecting every few seconds except of a few times it works flawlessly)

2) Right when it connects i power off the ESP32 with the switch and turn it back on.

By the time i do this it instantly shows the speedometer and battery percentage while being disconnected from the Bluetooth.
The controller DOES NOT "time out" after like 1 minute of riding like it usually would when the scooter is locked.

----> Scooter works great by powering on and off just the ESP32 by a switch.
Unlocked firmware without speed limit, greenbox version was 2.4.0.0.

----> Lights remain on after you switch the ESP32 OFF and time out by their self after a few hours, there is virtually no battery drain.

----> After i switch on the ESP32 the lights turn on and the scooter works as usually without needing to connect to bluetooth ever again.

Issues:
-Scooter times out with the ESP32 at the same time interval the lights go out when you turn the ESP32 off, if you let it sit for a few hours tou need to restart the ESP32 to get it running again (2-3 hours later maybe more).

-This can work like that as long as the battery isn't completely drained, but having the scooter charged once without riding it still showed 100% when used 3 months after and had great mileage aswell.

If you want something like what i say in the first post, TRY THIS.

Maybe this "glitch" is going to work for everyone.

john-leo93, 2025-05-07:
> ...because its not meant to be used like that. There is a proper solution already coded inside, use it instead of "glitching" around.

I can assure you it doesn't matter as i can't use it at all in any other way.

Bluetooth connection to ESP32 connects-disconnects every few seconds on my current phone rendering any attempt to keep the scooter unlocked useless, once i press the "unlock" button the scooter locks again and shows "scan to unlock" as the BT connection drops.
I don't have enough time to send the "unlock forever" command either, i have tried.

If i want to turn the lights on-off it would be better to directly wire them to a couple switches like the guys with the hacked greenbox do.

aIecxs, 2025-05-08:
Is Bluetooth working for you on a plain ESP32? Bluetooth requires load up to 500mA - How do you power ESP32?

GM3322, 2025-05-08:
> > ...because its not meant to be used like that. There is a proper solution already coded inside, use it instead of "glitching" around.
> 
> I can assure you it doesn't matter as i can't use it at all in any other way.
> 
> Bluetooth connection to ESP32 connects-disconnects every few seconds on my current phone rendering any attempt to keep the scooter unlocked useless, once i press the "unlock" button the scooter locks again and shows "scan to unlock" as the BT connection drops.
> I don't have enough time to send the "unlock forever" command either, i have tried.
> 
> If i want to turn the lights on-off it would be better to directly wire them to a couple switches like the guys with the hacked firmware that keep the greenbox do. 

Use any Bluetooth terminal to send the unlock and then unlockforever command, you may consider trying other phones, most people had success with samsung phones, while on Xiaomi (HyperOS) its buggy. 

It may be even the issue with ESP32 itself but i doubt that.

GM3322, 2025-05-08:
> > Is Bluetooth working for you on a plain ESP32? Bluetooth requires load up to 500mA - How do you power ESP32?
> 
> I haven't tried any other ESP32, the BT just bugs out on my current phone and used to bug more rarely on my previous phone.
> 
> I have plenty of power through a step down converter reducing the scooter's battery voltage to 5v
> 
> The ESP32 does the same thing out of the scooter even when powered directly by a USB charger or a powerbank, the step down converter connected to 12v battery etc.
> 
> 
> 
> > Use any Bluetooth terminal to send the unlock and then unlockforever command, you may consider trying other phones, most people had success with samsung phones, while on Xiaomi (HyperOS) its buggy.
> > 
> > It may be even the issue with ESP32 itself but i doubt that.
> I have tried doing that but the connection doesn't give me sufficient time.
> 
> With all the above said, it might be an issue with the ESP32 its self, or the fact that i've indeed used Xiaomi phones which appear to be buggy on BT even when i connect to other BT devices.
> Like when i go connect my car's radio and it bugs and shows it connects automatically to a device that isn't even nearby instead.
> 
> But still i'm used to scooters having a power button and auxillary phone app that isn't required, so i'd admire having the lime working like that.
> 
> So as long as the "glitch" i've found works reliably like it seems it does till now... Yeah.

May be the issue, if you have any chance of installing crom on any of your older phones (with AOSP) that may be helpfull, that said, try "Bluetooth Terminal" from Play Store and setup Unlock and Unlock Forever macros (commands are unlock and unlockforever) and set newline in settings to none if the ESP ignores your input. Let me know if that helped

aIecxs, 2025-05-08:
Works fine on Xiaomi phone here. what step down converter?

If you don't want Bluetooth unlock why are you using this code at first place? Write a mimimal code that only send unlock and heartbeat hex strings inba loop, use a switch for LOCK_PIN cable instead of switching on/off with microcontroller.