A-Emile/Lime_Gen3_IoT_Replacement #33: esp32 wroom not communicating with scooter

valentin2009-tech, 2025-04-01:
So I am using the esp wroom 32 and uploaded the code on the esp and when I plug it into the scooter controller, it keeps the scooter unlocked for a minute or two, and then it locks again. And when I try to unlock it from the application it doesn't work. Could someone help ?

dzsozineni, 2025-05-12:
I have the same problem

GM3322, 2025-05-12:
Did you flash "locked" firmware from scooterhacking repository? Does the battery percentage show up?

aIecxs, 2025-06-20:
Try with level shifter.

_Originally posted by @aIecxs in [#15](https://github.com/A-Emile/Lime_Gen3_IoT_Replacement/issues/15#issuecomment-2433761711)_
> The 3.3v HIGH Level is maybe not high enough for 5v HIGH level, so behavior is undefined.