A-Emile/Lime_Gen3_IoT_Replacement #34: Apple ios app

dzsozineni, 2025-04-04:
Hello,
Does anyone know a working app that can control an ESP32 board from an iOS device?
I found a picture of an iOS app on the internet last time, but unfortunately, I couldn’t find any source for it.

[image]

RazvanRex, 2025-04-04:
Hi, the screenshot is and old picture from my app. It is not public and i am not looking to share it with people now. Will _probably_ release it to the public sometime, but it is not ready now. 🙂

dzsozineni, 2025-04-04:
And is there any public app, even a CLI-based one, that works?😅

RazvanRex, 2025-04-04:
[LightBlue](https://apps.apple.com/ro/app/lightblue/id557428110) worked very great for me while testing